androidthings/contrib-drivers #1: Contributing new drivers

MacroYau, 2016-12-14:
Congratulations to the Android Things announcement, and thanks for providing this awesome repository!

I am exploring the Android Things platform with my Raspberry Pi 3 and gaining experience in developing drivers various I/O peripherals (so far, I am working on drivers for TI's INA219 current sensor and the environmental sensors on the Raspberry Pi Sense HAT). Will this repository accept drivers from individual developers, or should we host them on our own? In the long run, are there any plans for Google to offer a centralized repository (community-contributed or otherwise) for drivers of common peripherals?

Thanks and keep up the good work!

cdut007, 2016-12-14:
google  can write driver ,os ,kernel everything in java code ?  let's change the world by openJDK

jdkoren, 2016-12-14:
Hi @MacroYau, we definitely welcome contributions from external developers! For new features and fixes on existing drivers, refer to the [CONTRIBUTING.md](https://github.com/androidthings/contrib-drivers/blob/master/CONTRIBUTING.md) file. For new drivers, in general we suggest to host it on your own repo. We do have future plans to index drivers, both ours and from third-party developers.

Specifically for the Pi Sense HAT, we have a very early draft implementation. If you want, we can push it to a development branch here and take your contributions.

Happy building!

MacroYau, 2016-12-15:
@jdkoren Thanks for your information!

Of course, it would be great if you can share the Sense HAT driver working draft with us.

proppy, 2016-12-17:
@MacroYau feel free to take a look at https://github.com/proppy/contrib-drivers/tree/sensehat/sensehat

proppy, 2016-12-17:
Closing this issue, let's iterate on this on the branch I just linked.

closed 2016-12-17 by proppy