androidthings/contrib-drivers #2: USB diver access on raspberry pi 3

PratikGajera, 2016-12-16:
Hello Team,

Successfully flash iot_rpi3.img image in raspberrypi 3. Now i want to access USB device.But i didn't know that how can access usb device over Android Things OS in raspberryPi 3.
Please Help me as soon as possible!!

proppy, 2016-12-17:
@PratikGajera As mentioned in the [release notes](https://developer.android.com/things/preview/releases.html) USB APIs are currently disabled in the Developer Preview RC1.

I would encourage you to track this [feature request](https://code.google.com/p/android/issues/detail?id=230305) on b.android.com to track the progress around this. (This repository is used to release user-space contrib-drivers)

closed 2016-12-17 by proppy

PratikGajera, 2016-12-19:
Thank you proppy.