androidthings/contrib-drivers #4: Missing LEDSTRIP_LENGTH constant

plattysoft, 2016-12-19:
IT appears on the sample code, but it is not present on the code.

jdkoren, 2016-12-19:
@plattysoft Just to clarify, this is for the APA102 driver?

proppy, 2016-12-19:
I Think @plattysoft is referring to the RainbowHat snippets:
https://github.com/androidthings/contrib-drivers/tree/master/rainbowhat#sample-usage

They mention `RainbowHat.LEDSTRIP_LENGTH]` but that doesn't exist yet in the driver.

plattysoft, 2016-12-19:
Yes, that is it. It is just a constant with value 7, very simple to fix.

proppy, 2016-12-19:
@plattysoft yep, [contributions](https://github.com/androidthings/contrib-drivers/blob/master/CONTRIBUTING.md) are welcome!

ipereziriarte, 2016-12-23:
@proppy I've created a small patch for the issue.
https://github.com/androidthings/contrib-drivers/pull/7

proppy, 2017-01-10:
Fixed with e3a4cff6cfa4504f4647b29a4402eb3e77235ca1

closed 2017-01-10 by proppy