androidthings/contrib-drivers #15: [RainbowHat] openBuzzer() not available

StefMa, 2017-01-03:
The latest [commit](https://github.com/androidthings/contrib-drivers/blob/efdbd9cb84895ea0cb1414c258f696c55afe8688/) from the RainbowHat driver say something about `openBuzzer()` inside the [`README.md`](https://github.com/androidthings/contrib-drivers/blob/efdbd9cb84895ea0cb1414c258f696c55afe8688/rainbowhat/README.md).

Unfortunately isn't there any `openBuzzer()` method inside the [`RainbowHat` class](https://github.com/androidthings/contrib-drivers/blob/efdbd9cb84895ea0cb1414c258f696c55afe8688/rainbowhat/src/main/java/com/google/android/things/contrib/driver/rainbowhat/RainbowHat.java) but a `openPiezo()`

What should we do now?
Change the `README` or method? 🤔

jdkoren, 2017-01-09:
We have this fixed internally, will close once the update is published.

proppy, 2017-01-18:
Fixed with 61c5a0ae879379d91d0fa5ae4e432d42ea3ef99c

closed 2017-01-18 by proppy