androidthings/contrib-drivers #31: [RainbowHat] Apa102 not behaving as expected

loeschg, 2017-01-17:
I'm not sure if this is a bug or not, but I'm seeing the following behavior when using the Apa102 led strip:

I'm trying to turn the pixels/LEDs on when pressing one of the buttons and off after releasing:

```
    private void doRainbow(boolean pressed) {
        Log.d(TAG, "rainbow pressed: " + pressed);
        ledstrip.setBrightness(1);
        try {
            int[] rainbow = new int[NUM_LEDS_IN_STRIP];
            if (pressed) {
                for (int i = 0; i < rainbow.length; i++) {
                    rainbow[i] = Color.HSVToColor(255, new float[]{i * 360.f / rainbow.length, 1.0f, 1.0f});
                }
            } else {
                for (int i = 0; i < rainbow.length; i++) {
                    rainbow[i] = Color.HSVToColor(0, new float[]{0, 0, 0});
                }
            }
            Log.d(TAG, "rainbow being written: " + rainbow[0] + " " + rainbow[1] + " " + rainbow[2] + " " + rainbow[3] + " " + rainbow[4] + " " + rainbow[5] + " " +rainbow[6]);
            ledstrip.write(rainbow);
//            ledstrip.write(rainbow); <--- if I write 2x, this all works as expected
        } catch (IOException e) {
            e.printStackTrace();
        }
    }
```
This doesn't seem to work. It'll end up reversing what I want to have happen (will be on when not pressed even though the log indicates I just wrote [0,0,0,0,0,0,0] to the ledstrip).

If I write 2x, it works as expected. Am I misusing this, or is it a bug?

proppy, 2017-01-18:
Will try to reproduce with the bare `Apa102` driver.

tomaszrykala, 2017-01-25:
I think I am observing the same behaviour - when I set colors to the strip, they start lighting up with a step delay https://github.com/tomaszrykala/DoReFindMi/issues/3

IvanWooll, 2017-02-16:
I'm experiencing the same thing however I don't think it's the Apa102 that's the problem.
I have a button press method containing two method calls, one after the other:
The first one clears the AlphaNumericDisplay;
`    private void clearText() {
        AlphanumericDisplay segment = null;
        try {
            segment = RainbowHat.openDisplay();
            segment.setEnabled(false);
            segment.close();
        } catch (IOException e) {
            e.printStackTrace();
        }
    }`
The second one writes a colour to the Apa102;
` private void lightLedStrip(float[] hsvColor) {
        Apa102 ledstrip;
        try {
            ledstrip = RainbowHat.openLedStrip();
            ledstrip.setBrightness(1);
            int[] colors = new int[RainbowHat.LEDSTRIP_LENGTH];
            for (int i = 0; i < colors.length; i++) {
                colors[i] = Color.HSVToColor(hsvColor);
            }
            ledstrip.write(colors);
            ledstrip.close();
        } catch (IOException e) {
            e.printStackTrace();
        }
    }`

A button press will call one of these two methods per turn.

loeschg, 2017-03-21:
I believe this is likely the problem we're seeing: http://blog.pimoroni.com/apa102-variants/

>It turned out that the solution was just to increase the number of clock cycles in the end frame to the magic number of 36 (!?), and everything behaved as expected.

The specific commit for their python lib - https://github.com/pimoroni/blinkt/commit/829aac0106bb08b6d10bba5c3ad001b468622c39.

While messing around with the apa102 on another platform, I discovered writing an additional 4 bytes (a full LED frame) before the closing sequence (1*32) made everything work for me. 

[image]
https://cdn-shop.adafruit.com/datasheets/APA102.pdf

proppy, 2017-03-21:
@loeschg +1 for diagnosing the issue!

loeschg, 2017-03-21:
I should add that my solution isn't perfect. I end up getting a brief flash on the rightmost led (I assume due to whatever my additional led frame gets evaluated as). If I figure out a better solution, will post back here.

mangini, 2017-06-15:
This should be fixed in apa102 driver version 0.4. Please verify.

closed 2017-06-15 by mangini

proppy, 2017-06-15:
Fixed with https://github.com/androidthings/contrib-drivers/commit/f8cd83c005ca70d97ed1c60f35e71adcf5cc9276

Thanks for the diagnosis @loeschg!