androidthings/contrib-drivers #35: Speaker does not have play(tone, duration) method

tomaszrykala, 2017-01-25:
The readme for PWM Speaker describes an overloaded `play()` method that takes a tone and a duration as arguments. The driver however, supplies only a method that allows the tone.

readme source:
https://github.com/androidthings/contrib-drivers/tree/master/pwmspeaker

extract:
```java
// Make it play:

try {
    mSpeaker.play(440 /* tone */, 3000L /* duration */);
} catch (IOException e) {
    // error setting speaker
}
```

The only currently existing method:
```java
    public void play(double frequency) throws IOException, IllegalStateException {
...
    }
```

proppy, 2017-01-26:
Sounds like a documentation bug :)

tomaszrykala, 2017-01-26:
Can update the documentation to address this, but is there somehow we could expand on this, admiteddly not ideal, solution? https://github.com/androidthings/contrib-drivers/pull/36

tomaszrykala, 2017-01-26:
Thank you for the input. I will update the documentation in the meantime.

mangini, 2017-01-27:
@tomaszrykala Supporting duration essentially means having a sequencer in a separate thread that handles the timers, so it's not trivial. We have an initial implementation in our internal code review. Can you open a separate "enhancement" issue so we can keep track of it? Thanks!

tomaszrykala, 2017-01-28:
@mangini Here it is. https://github.com/androidthings/contrib-drivers/issues/38 Thanks.

tomaszrykala, 2017-02-07:
This is closed by https://github.com/androidthings/contrib-drivers/pull/37

closed 2017-02-07 by tomaszrykala