androidthings/contrib-drivers #40: Bmx280 Driver I2C address

macdidi5, 2017-02-08:
Bmx280 class define 0x77 for I2C address, but bmx280 can config two I2C address, 0x76 and 0x77.

jvrodley, 2017-02-21:
Confirming - I have a BME280 from here: 
https://www.amazon.com/gp/product/B01LETIESU/ref=oh_aui_search_detailpage?ie=UTF8&psc=1

Responds on 76, not 77.

BaN4NaJ0e, 2017-02-28:
Mine responds on 76 too. Would be nice to have an option to configure the address.

acristescu, 2017-03-03:
Here is a workaround for this (that also enables humidity sensor readings if you happen to be on the BME280). There is full code over there, or you can just grab the part that enables an arbitrary address:
http://zenandroid.io/writing-a-driver-for-android-things-bme280-humidity-sensor/

Btw, I would consider contributing the code back in this repo, but I don't know how to :)

jdkoren, 2017-03-03:
@acristescu Thanks for looking into this and writing about it :)

If you'd like to contribute changes to this repo, [`CONTRIBUTING.md`][1] describes the process. You would make a fork, apply your changes there, and then open a pull request back to this repo.

[1]: https://github.com/androidthings/contrib-drivers/blob/master/CONTRIBUTING.md

macdidi5, 2017-03-06:
if you use BMP280 module, connect SD0 to 3.3v, can config i2c address to 0x77. 
[image]

mangini, 2017-06-15:
Fixed in bmx280 driver version 0.3. Now there's an optional constructor argument that takes the I2C address. Default is the same as before.

closed 2017-06-15 by mangini

proppy, 2017-06-15:
Fixed with https://github.com/androidthings/contrib-drivers/commit/e99264b32a29848256a15eb9e3fc13409d3d5598