androidthings/contrib-drivers #44: Implement BLE methods

Lemac13, 2017-04-06:
I can't find if android things have BLE enabled for RPI3.

Somebody Know if is it enabled?

Suggestion to implement the function for control BLE easily

Thank you

MacroYau, 2017-04-06:
It has been a [known issue](https://developer.android.com/things/preview/releases.html) that Bluetooth is not available in Android Things as of the DP2 release.

hammmmy, 2017-04-06:
I am also eagerly waiting for the release of the peripheral and central GATT apis to work on android things :)

mangini, 2017-04-06:
Bluetooth (classic and BLE) support has just been launched on DP3: https://android-developers.googleblog.com/2017/04/android-things-developer-preview-3.html
We have also provided a GATT server sample: https://github.com/androidthings/sample-bluetooth-le-gattserver

closed 2017-04-06 by mangini