androidthings/contrib-drivers #52: HT16K33 - display() and clear() not working correctly

ckoessler, 2017-06-05:
I am trying the ht16k33 driver on a raspberry pi3 running preview 3 and the version 0.3 of the driver. My display is the Adafruit 1.2" 7-segment LED HT16K33 Backpack.

When sending the command clear(), the last digit on the display still light up some segments (cf picture)
[image]

I then ran some other test with different texts. Here are the results.

[image]
"XOXO"

[image]
"abcd"

Digits from "0000" to "9999"
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

proppy, 2017-06-05:
@ckoessler can you confirm how you wired the display to the Raspberry Pi?

ckoessler, 2017-06-05:
@proppy 
SCL -> Pin5 (SCL)
SDA -> Pin3 (SDA)
GND -> Pin6 (GND)
+5V -> Pin4 (5v)
V_IO -> Pin1 (3.3V)

ckoessler, 2017-06-06:
@proppy were you able to reproduce the problem?

proppy, 2017-06-06:
@ckoessler I don't have the part to test this yet, but I'd be curious if you've got the same issue with another `ht16k33` based peripherals (if you happen to have one).

ckoessler, 2017-06-06:
@proppy I don't have any other. I am curious as well and wondering if I just have a bad display. If this is the case I need to send it back ASAP...

proppy, 2017-06-06:
/cc @beriberikix who recently stole my `Adafruit 1.2" 7-segment LED HT16K33 Backpack`, can you try reproducing this issue with this sample:
https://github.com/androidthings/drivers-samples/tree/master/ht16k33

ckoessler, 2017-06-07:
@proppy I'm sending back my display and switching to another model. Hopefully one of you will be able to check if the code has a bug or not. Thanks!

beriberikix, 2017-06-18:
@proppy I don't have the an AThings device setup at the moment - do you want me to send the 7seg display to you?

beriberikix, 2017-06-19:
Just a thought - have you verified that the pins are providing enough current to drive the display?

ckoessler, 2017-06-19:
I only have one pi3 loaded with AT. In doubt, I assumed I had a bad display and sent it back before the end of the return window. I won't be able to test changes.

beriberikix, 2017-06-20:
I can confirm something isn't working correctly. I too saw weird behavior using the contrib-driver, while it worked perfectly fine with [code from Adafruit](https://github.com/adafruit/Adafruit_LED_Backpack/blob/master/examples/sevenseg/sevenseg.ino) on an Arduino.
[image]
[image]

jumoog, 2017-08-05:
I have the same problem with the Grove-LCD RBG Backlight

mangini, 2017-08-08:
@beriberikix Can you confirm that no exception was thrown? In the past I had weird characters showing up when my code threw an exception, probably because the whole process went down before finishing writing to the bus. Also, are you getting this issue consistently, or randomly?