androidthings/contrib-drivers #62: BMX280 Pressure only readings do not work

lgleasain, 2017-07-23:
When taking readings from a BMP280 sensor it only updates readings for pressure if you are also monitoring the temperature. If you try to only monitor the pressure no reading is ever returned.

jdkoren, 2017-07-24:
These peripherals require that temperature sampling be enabled because part of the pressure compensation formula uses the raw temperature reading. When you call `readPressure()`, if temperature sampling is disabled, it will throw an exception; otherwise it should take a sample and return a value. Is that the behavior you are seeing?

mangini, 2017-07-24:
I found this same issue earlier last week while implementing something else. The problem is that, since the pressure sampling requires temperature sampling, both need to have their oversampling enabled. Currently, the bmx280 pressure userdriver does not enable temperature oversampling, requiring the temperature userdriver to also be enabled. I've fixed it last week in an internal build and will be publishing it in the next few days.

lgleasain, 2017-07-24:
Thanks,  from a user perspective you needed to call registerPressureSensor() and register a listener which didn't seem right.  I was also thinking of taking a stab at the BME280 humidity sensor support, I'll open another issue for this to make it easier to track, but wanted to make sure that I'm not duplicating effort.

mangini, 2017-07-24:
That (bme280) is what I was working on :-)  It's being reviewed and should be out soon.

mangini, 2017-08-08:
Fixed on https://github.com/androidthings/contrib-drivers/commit/f859c093f44233ece7881703db26afc49b22f87b

closed 2017-08-08 by mangini