androidthings/contrib-drivers #63: Add support for BME280 Humidity readings

lgleasain, 2017-07-24:
I'm thinking of taking a stab at this, but wanted to open an issue so that we are not duplicating efforts.  This will be added to the BMX280, and add support for humidity along with error messages if a user tries to use an un-supported function on a sensor,  IE: trying to read humidity on a BMP-280 which does not support it. I'd also like to bubble up the device ID as part of the driver so that someone can determine which sensor is hooked up programatically. It looks like it's already in the driver but just needs to be exposed.

mangini, 2017-07-24:
Thanks for opening this bug, @lgleasain . I have the BME280 code under review internally and will be publishing it in the next few days.

lgleasain, 2017-07-25:
Cool!  Looking forward to it!

mangini, 2017-08-08:
Humidity support (bme280) submitted on https://github.com/androidthings/contrib-drivers/commit/2a23e0e3f438dfc4e76e727657c96765c1f7cb8a

closed 2017-08-08 by mangini