androidthings/contrib-drivers #71: Support for Adafruit PiTFT 2.2" HAT Mini Kit

jaumard, 2017-09-12:
Hello,
For a project I buy this display https://www.adafruit.com/product/2315 does it will work out of the box with android things ? If not what should I do to make it work ?

Fleker, 2017-09-12:
I have not tested out the functionality of this board. You may need to modify the `config.txt` file for the Pi in order for it to work.

jaumard, 2017-09-13:
When I receive it I'll try then :) I'll let you know if it's compatible out of the box but I see that there already a driver for an oled display so I suppose a driver will be needed too for this one :/

jaumard, 2017-09-16:
@Fleker Ok just receive it, didn't work out of the box, where can I find this config.txt and modify it please ?

Fleker, 2017-09-17:
Try looking at [this answer](https://stackoverflow.com/questions/43265564/how-do-i-make-the-pi-topceed-screen-work-with-android-things/43265704) although I don't know how to help specifically for your device

TwistedTransistor, 2017-11-30:
Shouldn't be too hard to do. It supports 4-wire 8-bit SPI communication.  As a starter project, you could do something as simple as reading a bitmap's pixels, 1 by 1, and pumping each pixel's RGB data over SPI to the device.

The make it more interesting, moving a mouse connected to the device, pickup the mouse's position and draw a pointer bitmap over the background bitmap at the mouse's reported position.

Looks like you can get pretty fancy with it the more you build into your driver.

mangini, 2017-12-19:
@jaumard This display has an SPI interface, so you need a user-space driver and it won't show up as a regular display to Android, which means you won't easily be able to draw regular Android UI on it. If you want to go that route, I recommend that you port [Adafruit's SPI driver](https://github.com/adafruit/Adafruit_ILI9340/blob/master/Adafruit_ILI9340.cpp):

It shouldn't be too hard. The Wiring API for SPI used by their driver can easily be mapped to the [Android Things SPI](https://developer.android.com/things/reference/com/google/android/things/pio/SpiDevice.html). You can either manually convert C++ to Java, or use [Android Things NDK API](https://github.com/androidthings/native-libandroidthings/blob/master/armeabi-v7a/include/pio/spi_device.h)

closed 2017-12-19 by mangini