androidthings/contrib-drivers #72: Support for HD44780 based LCD displays

piotrserafin, 2017-09-18:
Hello,

I'm working on HD44780 based LCD support in User-space drivers layer. I wish to share after when its completed.

robertoestivill, 2017-11-20:
@piotrserafin that would be cool. I got my hands in one of those and I'm looking for a driver or material to start digging into. Any tips?

piotrserafin, 2017-11-27:
Hi,

Im using only datasheet for controller which have be found here:
https://www.sparkfun.com/datasheets/LCD/HD44780.pdf

There is a lot of sources linked to wikipedia:
https://en.wikipedia.org/wiki/Hitachi_HD44780_LCD_controller

TwistedTransistor, 2017-11-30:
Should be very easy to write this driver.  You can probably use the Pi4J code to do most of it.

https://github.com/Pi4J/pi4j/blob/master/pi4j-example/src/main/java/LcdExample.java

piotrserafin, 2017-11-30:
Sure it is easy but the whole idea is to use android things sdk for that instead of ext. lib.

TwistedTransistor, 2017-11-30:
Oh I did not mean to use it as it is, but to use it as a reference to port it.

leinardi, 2017-12-21:
If someone is interested in a I2C driver for the HD44780, I just published one in jcentral.
Info and sources available here: https://github.com/leinardi/androidthings-drivers/tree/master/driver-pcf8574-hd44780