androidthings/contrib-drivers #80: APA102 on Intel Edison: Unknown I/O name SPI2

Zitt, 2017-11-26:
I'm writing my second Android app using AndroidThings... first being a blink demo. So admittedly; I'm likely out of my element.

When I run my application; I'm getting the following error on the following lines:
```
            Log.d(TAG, "Initializing LED strip");
            mApa102 = new Apa102(apa102BusName, LED_MODE);
```

where 
`private static final String apa102BusName = "SPI2";`

LogCat reports:
```
11-26 07:36:55.008 19031-19031/com.pinball_mods.zitt.edisontest D/HomeActivity: Initializing LED strip
11-26 07:36:55.019 19031-19031/com.pinball_mods.zitt.edisontest I/HomeActivity: Unable to configure SPI2 for APA102 led strip
                                                                                com.google.android.things.pio.PioException: android.os.ServiceSpecificException: Unknown I/O name SPI2
...
```

Do I have a broken Edison build chain? What am I missing

Zitt, 2017-11-26:
More debugging shows that 
```
PeripheralManagerService service = new PeripheralManagerService();       

        List<String> deviceList = service.getSpiBusList();
        for(String dev : deviceList) {
            Log.i(TAG, dev);
        }
```

returns `SPI2.1` instead of `SPI2`.

Using that string instead of SPI2... still doesn't work; but fails in a different way:
```
11-26 08:39:41.245 1466-1466/com.pinball_mods.zitt.edisontest I/HomeActivity: Unable to configure SPI2.1 for APA102 led strip
                                                                              com.google.android.things.pio.PioException: android.os.ServiceSpecificException: SPI2.1 failed to apply the required pin mux
                                                                                  at com.google.android.things.pio.SpiDeviceImpl.<init>(SpiDeviceImpl.java:41)
```

Fleker, 2017-11-28:
Looks like the Edison is using [those pins](https://developer.android.com/things/hardware/edison-arduino-io.html) for multiple things. There may be a way to configure the pin muxing on the system, but since that platform is deprecated it may not be high priority.

Are you using IO10-13 for anything else? Do you have any other apps currently installed?

Zitt, 2017-12-02:
Sorry; missed this update; received no email.

I am using IO13 for the LED blink as a heartbeat indicator.
I commented out that code; and I'm not getting any exceptions. I'm going to wire a stip up and see if it works.

Zitt, 2017-12-02:
Good call. IO13 / led blink was the cause.
IO13 shares the LED with one of the SPI pins on Intel Edision which was causing all of the issues.
Closing this as not a bug.

closed 2017-12-02 by Zitt