androidthings/contrib-drivers #81: Approval to contribute?

TwistedTransistor, 2017-11-29:
Hello, guys.

I have written a driver for the MAX31865 RTD-to-Digital converter.  It is limited at this point in that you have to ask the device to make a measurement for you, but it works.  I am currently adding code in order to take full advantage of the device's capabilities, including repeated automatic measurements at this point.

I would like to contribute to this project and add that driver here.

I followed the instructions here (https://github.com/androidthings/contrib-drivers/blob/master/CONTRIBUTING.md) and submitted the individual CLA, but have yet to hear anything back.

Is there anything else I need to do before I can start contributing to the project?

Thanks.

-Mario

Fleker, 2017-11-30:
We're still trying to figure out the process for accepting new drivers to the repository, as there's not a clear understanding of who would own the maintenance of this driver over time.

If you are writing a driver for a new peripheral, you can post it on your own GitHub and add it to JCenter or similar services on your own.

TwistedTransistor, 2017-11-30:
Understood.  I can put it in a repository of my own.  Once I do that, what are the possibilities you guys can create a repository for it under Contrib-Drivers with just a readme file that forwards people to my repository.  That way it would at least make people who come looking for a driver here to find it in my repository.

Thanks.

Fleker, 2017-12-02:
We are thinking about better ways to show drivers.

closed 2017-12-02 by Fleker