androidthings/contrib-drivers #82: I would like to contribute with the driver for the SH1106

leinardi, 2017-12-03:
Hi,

as requested in the [contributing](https://github.com/androidthings/contrib-drivers/blob/master/CONTRIBUTING.md) I'm opening this issue to propose the addition of a new driver I wrote for the SH1106 OLED display.

The driver is based on the already available SSD1306 driver and the code can be found here: https://github.com/leinardi/androidthings-kuman-sm9/tree/master/sh1106

Please let me know if you are interested in a pull request.

Fleker, 2017-12-03:
Right now we're not accepting external contributions for new drivers to this repository.

closed 2017-12-03 by Fleker

leinardi, 2017-12-15:
In case someone is interested in this driver, is now available on JCenter, more info here: https://github.com/leinardi/androidthings-drivers