androidthings/contrib-drivers #84: Voicehat support for NXP i.MX 7Dual

rachitmishra, 2017-12-16:
Hi, Is it possible to use voicehat driver with Pico developer kit?

Fleker, 2017-12-16:
By default, [the imx7 I/O](https://developer.android.com/things/hardware/imx7d-pico-io.html) does not support I2S, it's on a different set of pins. The VoiceHat peripherals, such as the LED and button, can be opened using the `VoiceHat.openLed(pinname)` and `VoiceHat.openButton(pinname)` methods.

Technically, yes you can use the VoiceHat driver with the imx7.

rachitmishra, 2017-12-20:
yup, I was wondering if i can get the right set of pings to open, i think i can control the mic as well as the speaker. thank you 👍

Fleker, 2017-12-20:
You can see some of the matches in the [Google Assistant sample](https://github.com/androidthings/sample-googleassistant/blob/master/app/src/main/java/com/example/androidthings/assistant/BoardDefaults.java).

rachitmishra, 2017-12-28:
Great! Thanks, I was looking at the python samples :'D

closed 2017-12-29 by Fleker