androidthings/contrib-drivers #85: Gradle not resolving after compiling rainbow hat

soorajshankar, 2017-12-17:
Error:Execution failed for task ':app:preDebugAndroidTestBuild'.
> Conflict with dependency 'com.android.support:support-annotations' in project ':app'. Resolved versions for app (24.2.0) and test app (25.4.0) differ. See https://d.android.com/r/tools/test-apk-dependency-conflicts.html for details.

Fleker, 2017-12-17:
Can you post the line in your gradle file you're using to import the
rainbow hat?

soorajshankar, 2017-12-17:
I fixed that by removing expresso from dependency .
But seems some version mismatching is there.
BTW this was the way I imported rainbowhat
>     compile 'com.google.android.things.contrib:driver-rainbowhat:+'
also tried with older version 0.7 and all 

> Min SDK version 26
> target SDK version 26 (tried with 24 as min also then things was not compatible)

Fleker, 2017-12-17:
Looks like Rainbow Hat uses [24.2.0 of Android Support Annotations](https://github.com/androidthings/contrib-drivers/blob/master/rainbowhat/build.gradle#L32), but only for testing. I don't know why it would conflict with your version.

soorajshankar, 2017-12-18:
I think Android Support Annotations are defined as **Implementation** but not **Test Implementation**

I doubt that could be a problem if I use another support library with different version

Fleker, 2017-12-18:
Yes you're right it isn't defined as **testImplementation**, but I do wonder why this library is using it. It probably should be updated or perhaps removed.

But **implementation** shouldn't pass the dependencies through AFAIK.