androidthings/contrib-drivers #90: NMEAgpsDriver

prasha-findmind, 2018-02-28:
my program crashes when i try to register the nmeagpsdriver can someone please help me

Fleker, 2018-02-28:
Have you tried to reboot the device after installing your app? Maybe it is related to permissions.

prasha-findmind, 2018-03-01:
[image]

that line corresponds to the UserDeviceDriver.register();

prasha-findmind, 2018-03-01:
@Fleker 
Yes i tried it still crashes

prasha-findmind, 2018-03-01:
The NMEAGpsModule works fine I'm having issues only with the driver

Fleker, 2018-03-01:
Do you have the permission `<uses-permission android:name="com.google.android.things.permission.MANAGE_GPS_DRIVERS" />`

prasha-findmind, 2018-03-01:
Yeah I have included that in the manifest file

Fleker, 2018-03-01:
I'm not sure what else it could be. Can you post your stacktrace error?

prasha-findmind, 2018-03-01:
java.lang.RuntimeException: Unable to create service com.example.acer.integrated.GpsService: java.lang.SecurityException: Caller lacks required permission com.google.android.things.permission.MANAGE_GPS_DRIVERS

prasha-findmind, 2018-03-01:
I'm getting this line in the error, but I have included that line in the manifest file, I.m sorry I don't know how to post the stacktrace error

Fleker, 2018-03-01:
Can you post the contents of your manifest file? The permission tag may be in the wrong spot.

prasha-findmind, 2018-03-01:
<?xml version="1.0" encoding="utf-8"?>
<manifest xmlns:android="http://schemas.android.com/apk/res/android"
    package="com.example.acer.integrated">
    <uses-permission android:name="com.google.android.things.permission.MANAGE_GPS_DRIVERS" />
    <uses-permission android:name="android.permission.INTERNET"/>
    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
    <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
       <application>
        <uses-library android:name="com.google.android.things" />

        <activity android:name=".MainActivity">
            <intent-filter>
                <action android:name="android.intent.action.MAIN" />

                <category android:name="android.intent.category.LAUNCHER" />
            </intent-filter>
            <intent-filter>
                <action android:name="android.intent.action.MAIN" />

                <category android:name="android.intent.category.IOT_LAUNCHER" />
                <category android:name="android.intent.category.DEFAULT" />
            </intent-filter>
        </activity>
        
        <meta-data
            android:name="com.google.android.geo.API_KEY"
            android:value="@string/google_maps_key" />

        <activity
            android:name=".MapsActivity"
            android:label="@string/title_activity_maps"></activity>
        <service android:name=".GpsService"
            android:enabled="true"
            />
    </application>

prasha-findmind, 2018-03-01:
<?xml version="1.0" encoding="utf-8"?>
<manifest xmlns:android="http://schemas.android.com/apk/res/android"
    package="com.example.acer.integrated">
    <uses-permission android:name="com.google.android.things.permission.MANAGE_GPS_DRIVERS" />
    <uses-permission android:name="android.permission.INTERNET"/>
    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
    <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
       <application>
        <uses-library android:name="com.google.android.things" />

        <activity android:name=".MainActivity">
            <intent-filter>
                <action android:name="android.intent.action.MAIN" />

                <category android:name="android.intent.category.LAUNCHER" />
            </intent-filter>
            <intent-filter>
                <action android:name="android.intent.action.MAIN" />

                <category android:name="android.intent.category.IOT_LAUNCHER" />
                <category android:name="android.intent.category.DEFAULT" />
            </intent-filter>
        </activity>
        
        <meta-data
            android:name="com.google.android.geo.API_KEY"
            android:value="@string/google_maps_key" />

        <activity
            android:name=".MapsActivity"
            android:label="@string/title_activity_maps"></activity>
        <service android:name=".GpsService"
            android:enabled="true"
            />
    </application>

</manifest>

prasha-findmind, 2018-03-01:
[image]

Fleker, 2018-03-01:
So you have done the following, in this order:

* Added the permission to your app
* Installed/ran the app on your device
* Rebooted the device
* Reran the app on your device

And you're still seeing a security issue? What developer preview is running on the device, and what device is it?

Fleker, 2018-03-01:
At the moment rebooting is the current way to solve this while in development mode.

prasha-findmind, 2018-03-27:
I tried connecting a GPS via USB, to the Pi. and I checked the available UART ports 
onCreate: [MINIUART, UART0, USB1-1.5:1.0]
these were the available UART ports.
So I tried registering the third UART port as the NMEAGPSDriver's UART port("USB1-1.5:1.0"),
It returned a buffer overflow exception. Can I solve this somehow?

prasha-findmind, 2018-03-27:
03-27 07:23:14.741 2072-2072/? I/main: type=1400 audit(0.0:190): avc: denied { write } for name="trace_marker" dev="tracefs" ino=47 scontext=u:r:zygote:s0 tcontext=u:object_r:debugfs_tracing:s0 tclass=file permissive=1
03-27 07:23:14.751 2072-2072/? I/main: type=1400 audit(0.0:191): avc: denied { open } for path="/sys/kernel/debug/tracing/trace_marker" dev="tracefs" ino=47 scontext=u:r:zygote:s0 tcontext=u:object_r:debugfs_tracing:s0 tclass=file permissive=1
03-27 07:23:14.807 2072-2072/? I/zygote: Late-enabling -Xcheck:jni
03-27 07:23:15.329 2072-2072/com.example.acer.uart I/InstantRun: starting instant run server: is main process
03-27 07:23:15.571 2072-2072/com.example.acer.uart D/ContentValues: onCreate: [MINIUART, UART0, USB1-1.5:1.0]
03-27 07:23:15.748 2072-2072/com.example.acer.uart D/vndksupport: Loading /vendor/lib/hw/android.hardware.graphics.mapper@2.0-impl.so from current namespace instead of sphal namespace.
03-27 07:23:16.089 2072-2072/com.example.acer.uart W/NmeaParser: Checksum missing from incoming message
03-27 07:23:16.133 2072-2072/com.example.acer.uart W/NmeaParser: Checksum missing from incoming message
03-27 07:23:16.133 2072-2072/com.example.acer.uart I/chatty: uid=10030(com.example.acer.uart) identical 2 lines
03-27 07:23:16.133 2072-2072/com.example.acer.uart W/NmeaParser: Checksum missing from incoming message
03-27 07:23:17.007 2072-2072/com.example.acer.uart D/AndroidRuntime: Shutting down VM
03-27 07:23:17.016 2072-2072/com.example.acer.uart E/AndroidRuntime: FATAL EXCEPTION: main
                                                                     Process: com.example.acer.uart, PID: 2072
                                                                     java.lang.IllegalArgumentException: Invalid ASCII hex byte
                                                                         at com.google.android.things.contrib.driver.gps.NmeaParser.getHexDigit(NmeaParser.java:424)
                                                                         at com.google.android.things.contrib.driver.gps.NmeaParser.convertAsciiByte(NmeaParser.java:408)
                                                                         at com.google.android.things.contrib.driver.gps.NmeaParser.validateChecksum(NmeaParser.java:122)
                                                                         at com.google.android.things.contrib.driver.gps.NmeaParser.processMessageFrame(NmeaParser.java:75)
                                                                         at com.google.android.things.contrib.driver.gps.NmeaGpsModule.handleFrameEnd(NmeaGpsModule.java:207)
                                                                         at com.google.android.things.contrib.driver.gps.NmeaGpsModule.processBuffer(NmeaGpsModule.java:176)
                                                                         at com.google.android.things.contrib.driver.gps.NmeaGpsModule.readUartBuffer(NmeaGpsModule.java:161)
                                                                         at com.google.android.things.contrib.driver.gps.NmeaGpsModule.access$000(NmeaGpsModule.java:35)
                                                                         at com.google.android.things.contrib.driver.gps.NmeaGpsModule$1.onUartDeviceDataAvailable(NmeaGpsModule.java:139)
                                                                         at com.google.android.things.pio.UartDeviceImpl$UartDeviceCallbackDispatch.dispatchInterruptEvent(UartDeviceImpl.java:250)
                                                                         at com.google.android.things.pio.CallbackDispatch.onFileDescriptorEvents(CallbackDispatch.java:149)
                                                                         at android.os.MessageQueue.dispatchEvents(MessageQueue.java:284)
                                                                         at android.os.MessageQueue.nativePollOnce(Native Method)
                                                                         at android.os.MessageQueue.next(MessageQueue.java:325)
                                                                         at android.os.Looper.loop(Looper.java:142)
                                                                         at android.app.ActivityThread.main(ActivityThread.java:6494)
                                                                         at java.lang.reflect.Method.invoke(Native Method)
                                                                         at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:438)
                                                                         at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:807)
03-27 07:23:17.031 2072-2072/com.example.acer.uart I/Process: Sending signal. PID: 2072 SIG: 9
a

Fleker, 2018-03-27:
This is an unrelated issue relating to the usage of the driver.

`java.lang.IllegalArgumentException: Invalid ASCII hex byte`

prasha-findmind, 2018-03-28:
Do you have any idea what might be going wrong?/

Fleker, 2018-03-28:
I'm not sure immediately what's going on. What GPS module are you connecting? Is it an NmeaGpsModule?

prasha-findmind, 2018-03-29:
Yeah it is an NMEA Gps module , it connects via the USB

prasha-findmind, 2018-04-27:
What all NMEA 2.0 sentences does this driver use to figure out the location?
($GPRMC , $GPRMB... etc )

devunwired, 2018-06-19:
Are you still experiencing issues with your GPS module? If so, please provide a link to the datasheet of the device you are using. The stack trace provided indicates it's not sending properly encoded NMEA data (or additional data the driver doesn't recognize).

prasha-findmind, 2018-06-20:
NO, I'm not facing the issue now, I was using a neo-8M GPS Module. But I think the driver was written for neo-6M GPS modules.

closed 2018-06-20 by prasha-findmind