androidthings/contrib-drivers #91: Support for Stepper Motors

IeiuniumLux, 2018-03-03:
Will [MotorHat](https://github.com/androidthings/contrib-drivers/tree/master/motorhat) or other contrib-driver have support for stepper motors?

MotorHat was inspired by Zugaldia's code; who already added support for stepper motors in his [repository](https://github.com/zugaldia/adafruit-motor-hat/blob/master/AdafruitMotorHat/lib/src/main/java/com/zugaldia/adafruit/motorhat/library/AdafruitStepperMotor.java).  Should we use that one instead?

Fleker, 2018-03-03:
I have a DC motor at my desk, but haven't looked into adding stepper motors to any project so far.

CavHaunch, 2018-10-25:
Could you suggest some resources for getting a project like that going on Android Things? What is the best way to control the power source to a motor from a microcontroller?

Fleker, 2018-10-25:
If you have a motor, you can use a motor driver, like an h-bridge, to get enough current from your microcontroller to the motor.

PaulTR, 2018-10-25:
It depends on the type of motor you have, I'm sure, but I had a lot of luck
porting this library over
https://github.com/tyhenry/CheapStepper/blob/master/CheapStepper.cpp for
the 28byj-48 ULN2003 5V stepper motor and driver board if you're just
looking for something to use.

fail572872916, 2018-11-09:
Android things's library is too small.