androidthings/contrib-drivers #92: APA102 Driver for WS2812b

prasha-findmind, 2018-03-06:
Can I use the APA102 Driver for WS2812b?
What are the modifications that should be done to use it?

Ic-ks, 2018-03-06:
I wrote a driver for these LEDs: https://github.com/Ic-ks/contrib-drivers/tree/master/ws2812b  I did already a pull request (https://github.com/androidthings/contrib-drivers/pull/76)

prasha-findmind, 2018-03-07:
the jcentre page is not opening, so I am not able to figure out the latest version, can you please let me know the latest version

prasha-findmind, 2018-03-07:
https://bintray.com/google/androidthings/contrib-driver-ws2812b/_latestVersion

it says this link is not valid I get a page not found error can you do something about it?

prasha-findmind, 2018-03-07:
I also tried searching in https://bintray.com/google/androidthings this link I couldn't find your driver, so please help me out

Ic-ks, 2018-03-12:
Sorry for the late response. You can import the driver by cloning the repository (https://github.com/Ic-ks/contrib-drivers.git) and using the import function of Android Studio in your project: File->New->Import-Module. Then select <CLONE_DIR>/contrib-drivers/ws2812b. The import function will copy ws2812 driver to your project. If you don't want a copy of the driver you could create a relative path dependency like described here: https://stackoverflow.com/questions/24975745/gradle-dependencies-compile-project-by-relative-path

closed 2018-03-27 by prasha-findmind