androidthings/contrib-drivers #95: Change Ht16k33 I2C_ADDRESS

IeiuniumLux, 2018-03-31:
[Ht16k33](https://github.com/androidthings/contrib-drivers/blob/df63c830fc3ebecf49769871c389aa02a60a72fc/ht16k33/src/main/java/com/google/android/things/contrib/driver/ht16k33/Ht16k33.java) is currently setting the I2C_ADDRESS to 0x70; which causes a conflict when you stack the RainbowHat and the Adafruit Motor Hat together because the latter responds to [address 0x70](https://learn.adafruit.com/adafruit-dc-and-stepper-motor-hat-for-raspberry-pi/stacking-hats#addressing-the-hats) - regardless of the address jumper settings or the value in the motor library/driver.

As a result, [AlphanumericDisplay](https://github.com/androidthings/contrib-drivers/blob/df63c830fc3ebecf49769871c389aa02a60a72fc/ht16k33/src/main/java/com/google/android/things/contrib/driver/ht16k33/AlphanumericDisplay.java) and [MotorHat](https://github.com/androidthings/contrib-drivers/blob/master/motorhat/src/main/java/com/google/android/things/contrib/driver/motorhat/MotorHat.java) are clashing with each other.  Therefore, please change the I2C_ADDRESS in Ht16k33 class to some other value than 0x70.

jdkoren, 2018-04-02:
Thanks for reporting this @IeiuniumLux 

It seems it is possible to [change the I2C address][1] of your `ht16k33`. However, we don't want to require this of every person who uses one, so I think the right approach is to keep 0x70 as the default address and provide an alternate constructor that accepts a different address (similar to the MotorHat driver), so you can change the address of your `ht16k33` and continue to use it.

We have made this change internally and will close this issue when we've released another version of this driver.

[1]: https://learn.adafruit.com/adafruit-led-backpack/changing-i2c-address

mangini, 2018-04-16:
@IeiuniumLux we've [published](https://github.com/androidthings/contrib-drivers/commit/8994fdbb49c6b6360b01c9ca4daaae29cecbf589#diff-ea98269f982f07e8b17326752d55716d) the optional address change. It's in the [version 0.6](https://github.com/androidthings/contrib-drivers/blob/master/ht16k33/CHANGELOG.md) of the ht16k33 driver.

closed 2018-04-16 by mangini

IeiuniumLux, 2018-04-16:
Excellent.  Thank you!