androidthings/contrib-drivers #103: Continuously report temperature - No sensor reported

paul-matthews, 2018-05-18:
Using the [code snippet for Continuously reporting temperature sensor](https://github.com/androidthings/contrib-drivers/tree/master/rainbowhat), no ambient temperature sensor is connected. However reading a single temperature value works fine. Is this library registering the driver for the Bmx280 on the board, because the sample doesn't cover it.

# Repro steps:
1) Copy code snippet into onCreate.
2) Run activity on a Pi 3 (with working other functions)

# Expected:
3) Observe logging of temperature sensor

# Actual:
3) Observed logging only includes: "I/SensorManager: DYNS Register dynamic sensor callback"
  - No events triggered into the SensorEventListener instance

On further inspection (attempting to use the getDefaultSensor() ) no sensors exist.

jdkoren, 2018-05-18:
It seems the snippet you linked to is missing a call to [`registerTemperatureSensor()`][1]. For now, please refer to the Activity code in the Bmx280 [driver sample][2], which demonstrates the same use case.

[1]: https://github.com/androidthings/contrib-drivers/blob/master/bmx280/src/main/java/com/google/android/things/contrib/driver/bmx280/Bmx280SensorDriver.java#L93
[2]: https://github.com/androidthings/drivers-samples/blob/master/bmx280/src/main/java/com/example/androidthings/driversamples/TemperatureActivity.java

jdkoren, 2018-05-21:
@plattysoft Not quite a duplicate (at least I don't think so), I think @paul-matthews issue is the code snippet is missing a call to `registerTemperatureSensor()`, so the sensor never actually becomes available.