androidthings/contrib-drivers #106: GPS NmeaParser & NmeaGpsModule VM Crash

FaznetSolutions, 2018-06-18:
Hi,

I am using Android Things v1.0.1 on the Raspberry Pi 3 with a Adafruit Ultimate GPS Breakout v3 and getting the following after running the GPS for a few minutes:

> 06-18 07:30:15.677 1590-1590/com.google.android.things.contrib.driver.gps D/AndroidRuntime: Shutting down VM
>     
>     
>     --------- beginning of crash
> 06-18 07:30:15.685 1590-1590/com.google.android.things.contrib.driver.gps E/AndroidRuntime: FATAL EXCEPTION: main
>     Process: com.google.android.things.contrib.driver.gps, PID: 1590
>     java.lang.ArrayIndexOutOfBoundsException: length=19; index=19
>         at com.google.android.things.contrib.driver.gps.NmeaParser.handleSatelliteData(NmeaParser.java:199)
>         at com.google.android.things.contrib.driver.gps.NmeaParser.processMessageFrame(NmeaParser.java:101)
>         at com.google.android.things.contrib.driver.gps.NmeaGpsModule.handleFrameEnd(NmeaGpsModule.java:206)
>         at com.google.android.things.contrib.driver.gps.NmeaGpsModule.processBuffer(NmeaGpsModule.java:175)
>         at com.google.android.things.contrib.driver.gps.NmeaGpsModule.readUartBuffer(NmeaGpsModule.java:160)
>         at com.google.android.things.contrib.driver.gps.NmeaGpsModule.access$000(NmeaGpsModule.java:41)
>         at com.google.android.things.contrib.driver.gps.NmeaGpsModule$1.onUartDeviceDataAvailable(NmeaGpsModule.java:138)
>         at com.google.android.things.pio.UartDeviceImpl$UartDeviceCallbackDispatch.dispatchInterruptEvent(UartDeviceImpl.java:250)
>         at com.google.android.things.pio.CallbackDispatch.onFileDescriptorEvents(CallbackDispatch.java:149)
>         at android.os.MessageQueue.dispatchEvents(MessageQueue.java:284)
>         at android.os.MessageQueue.nativePollOnce(Native Method)
>         at android.os.MessageQueue.next(MessageQueue.java:325)
>         at android.os.Looper.loop(Looper.java:142)
>         at android.app.ActivityThread.main(ActivityThread.java:6494)
>         at java.lang.reflect.Method.invoke(Native Method)
>         at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:438)
>         at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:807)
> 06-18 07:30:15.698 1590-1590/com.google.android.things.contrib.driver.gps I/Process: Sending signal. PID: 1590 SIG: 9

I do get the odd Invalid GSV Message logged but otherwise all messages are being received correctly and logged.  

Any ideas?

Thank you,

Paul

devunwired, 2018-06-19:
This was most likely due to a parsing bug in the driver code, which will be addressed in the next release of the driver.

FaznetSolutions, 2018-06-20:
I changed the Satellite parsing from 19 to 2 in NMEAParser and theres no more crashes.

```
// Parse the satellites in this message
for (int i = 4; i < 2; i += 4) {
    if (!nmea[i].isEmpty()) {
        Satellite sat = new Satellite();
        sat.svid = Integer.parseInt(nmea[i]);
        sat.elevation = parseTrackAngle(nmea[i+1]);
        sat.azimuth = parseTrackAngle(nmea[i+2]);
        sat.snr = parseSignal(nmea[i+3]);

        mSatellites.put(sat.svid, sat);
    }
}
```
I only tested it at 2, I haven't tried to see what the highest value is before it stops.

devunwired, 2018-06-26:
v1.1 of the GPS driver fixes parsing for variable length GSV messages.

closed 2018-06-26 by devunwired