androidthings/contrib-drivers #115: Raspberry pi ButtonInputDriver mutiple times

dionesxxx, 2018-11-07:
I am simulating a situation in raspberry pi where I call the register() method several times, approximately 300 times and after that number is generated an exception as below.

```
E/UncaughtException: java.lang.IllegalStateException: Fatal Exception thrown on Scheduler.
        at io.reactivex.android.schedulers.HandlerScheduler$ScheduledRunnable.run(HandlerScheduler.java:111)
        at android.os.Handler.handleCallback(Handler.java:790)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:164)
        at android.app.ActivityThread.main(ActivityThread.java:6494)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:438)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:807)
     Caused by: java.lang.IllegalArgumentException: Could not initialize input device.
        at android.os.Parcel.readException(Parcel.java:2008)
        at android.os.Parcel.readException(Parcel.java:1950)
        at com.google.android.things.userdriver.input.IInputDriverService$Stub$Proxy.createInputDevice(IInputDriverService.java:140)
        at com.google.android.things.userdriver.input.InputDriver.initialize(InputDriver.java:275)
        at com.google.android.things.userdriver.input.InputDriverManager.addInputDriver(InputDriverManager.java:48)
        at com.google.android.things.userdriver.UserDriverManager.registerInputDriver(UserDriverManager.java:106)
        at com.google.android.things.contrib.driver.button.ButtonInputDriver.register(ButtonInputDriver.java:93)
        at io.reactivex.internal.operators.maybe.MaybeObserveOn$ObserveOnMaybeObserver.run(MaybeObserveOn.java:104)
        at io.reactivex.android.schedulers.HandlerScheduler$ScheduledRunnable.run(HandlerScheduler.java:109)
        at android.os.Handler.handleCallback(Handler.java:790) 
        at android.os.Handler.dispatchMessage(Handler.java:99) 
        at android.os.Looper.loop(Looper.java:164) 
        at android.app.ActivityThread.main(ActivityThread.java:6494) 
        at java.lang.reflect.Method.invoke(Native Method) 
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:438) 
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:807) 
```

Do you know what might be happening? In all the calls I call the register and close them.

After running 300 times the raspberry does not respond to any other component that I try to use the ButtonInputDriver, only after the manual reset of raspberry the code returns to work

Fleker, 2018-11-07:
Are you adding any delays between opening/closing/opening? Perhaps there's some asynchronous logic that isn't ready by the next call.

dionesxxx, 2018-11-08:
Basically i did the code below inside **onCreate()** method.

```
        ButtonInputDriver buttonInputDriver;

        for (int i = 0; i <= 500; i++) {
            buttonInputDriver = new ButtonInputDriver(SENSOR_ENTRY, Button.LogicState.PRESSED_WHEN_HIGH, ENTRY);
            buttonInputDriver.setDebounceDelay(0);
            buttonInputDriver.register();
            buttonInputDriver.close();
        }
```

I also tried putting a time delay

```
        ButtonInputDriver buttonInputDriver;

        for (int i = 0; i <= 500; i++) {
            buttonInputDriver = new ButtonInputDriver(SENSOR_ENTRY, Button.LogicState.PRESSED_WHEN_HIGH, ENTRY);
            buttonInputDriver.setDebounceDelay(0);
            buttonInputDriver.register();
            buttonInputDriver.close();

            try {
                Thread.sleep(2000);
            } catch (InterruptedException e) {
                e.printStackTrace();
            }
        }
```

Both cases I have an IllegalArgumentException after 300 to 350 calls of the register() method.

Fleker, 2018-11-12:
This may be something to put in the [Android Things issue tracker](https://issuetracker.google.com/issues/new?component=192720&template=847005), as it may be more of a platform bug.

This probably shouldn't be done in a normal application though, for what it's worth.