androidthings/contrib-drivers #116: I have a device similar to i2c interface, but it is not a standard protocol like i2c. What should I do

jmying, 2018-11-19:
[image]

I need to control the clock and IO data myself

jmying, 2018-11-19:
How to independently control SCK and sdio on raspberry pi

Fleker, 2018-11-19:
The best option is to open a bunch of GPIO ports on the Pi and [bit-bang](https://en.wikipedia.org/wiki/Bit_banging) on the clock and data lines through software.

jmying, 2018-11-19:
There are examples?

jmying, 2018-11-20:
I think GPIO for java  can't do that. SDIO and SCLK  can be control by ndk ?

nohum, 2018-11-20:
@jmying The driver for TM1637 displays is bit-banging kind of a I2C protocol over custom GPIOs (using Java), I guess that is exactly what you'll want: https://github.com/androidthings/contrib-drivers/blob/master/tm1637/src/main/java/com/google/android/things/contrib/driver/tm1637/I2cBitBangDevice.java

jmying, 2018-11-20:
@nohum thanks ~

nohum, 2018-11-21:
Depending on what you mean with "delay" you could either
* sleep via `SystemClock.sleep(10)` (for e.g. 10ms) or
* you could register a `GpioCallback` via `PeripheralManager.getInstance().openGpio()` to be notified if your GPIO changes (and then do something with that information).

If you go with option one just pay attention that you do not sleep on the main thread as this will stall the whole app.