androidthings/contrib-drivers #118: [driver] Issue with driver for ADS1115

jenchenUA, 2018-12-12:
In case, when a range of measuring is set as RANGE_6_144V, and I try to measure 5v, I got ~4.8v instead of 5v.

Fleker, 2018-12-12:
There may be a margin of error that is listed in the device datasheet. There may also be a part of your circuit that is causing a small drop, such as a resistor.

jenchenUA, 2018-12-13:
> There may be a margin of error that is listed in the device datasheet. 

No, I've connected an input of the ADC directly to a source of 5v power without resistors and I've got only ~4.8v. But, there exists a strange thing that I've noticed, I've tried to measure 3.3v in such way and with my circuit, and it was measured correctly with the range: RANGE_6_144V.