androidthings/contrib-drivers #120: Button driver

YoeyHD, 2019-01-06:
Button presses are only recognized once. Only after a device reboot is a button press recognized again. 

I followed the sample code usage from the readme,  and inserted this into the onCreate() method to setup the button on device launch

    pButton = new Button(GPIO_P_BUTTON, Button.LogicState.PRESSED_WHEN_HIGH);

            pButton.setOnButtonEventListener(new Button.OnButtonEventListener() {
                @Override
                public void onButtonEvent(Button button, boolean pressed) {
                    doStuff();
                }
That code is obviously surrounded by a try/catch block.

I've added this to my module build.gradle file: 

    compile 'com.google.android.things.contrib:driver-button:1.0'

And this to my onDestroy method

    if(pButton != null){
                pButton.close();
            }

I tried different GPIO pins, different cables and different buttons. Always the same result.

Any idea what I could do to change that behavior?

PaulTR, 2019-01-06:
Any chance you can share an image of how things are plugged in?

YoeyHD, 2019-01-06:
I hope you can see everything you need to see with this

[image]

jdkoren, 2019-01-07:
@YoeyHD In your photo, your button seems to be lacking some connections.

With electrical switches (such as buttons), you need an additional connection on the output side (the side that goes to your GPIO) to "pull" the signal towards either power or ground, whichever is opposite the signal on the input side.

For example, take a look at the button on [this diagram](https://github.com/androidthings/sample-pio-app#build-and-install) from one of our samples. Notice the input side of the button connects directly to ground. On the output side, there are 2 connections: one to the GPIO, and one to 5V using a resistor. This is a [pull-up resistor](https://en.wikipedia.org/wiki/Pull-up_resistor). When the button is not being pressed, the GPIO signal is being _pulled up_ to 5V. (You could also swap 5V and ground to make a button that gets _pulled down_.)

Without a pull-up or pull-down resistor, the GPIO signal can "float" somewhere between high and low, which produces unintended behavior. Try connecting your button like the diagram and see if that helps. (This may not be your only issue, but it's worth fixing just so you can rule it out.)

YoeyHD, 2019-01-08:
It works! Thanks a lot! Since this is my first time with GPIO and everything, is there a way to trigger doStuff() once? Right now, it triggers more than once, usually twice. Sometimes even more, especially if I keep the button pressed for a longer period of time. The latter behavior wouldn't be such an issue. However, the former would be an issue

jdkoren, 2019-01-08:
@YoeyHD You can check the value of `pressed` in the callback and compare it to a previous value.

It also may depend on what resistor value you are using... I think most of the time I was using 10K resistors and typically didn't see a press register twice. Of course, that's not as foolproof as checking for a value change in the callback.

closed 2019-01-08 by jdkoren