androidthings/contrib-drivers #122: Support for UDP Sockets Android Things

nikiizvorski, 2019-02-14:
Hey guys i have reached a bit of desperation for hours and hours debugging. And i thought if i could reach out to the active community here for some help. I would like to ask if would be possible to have UDP Sockets on Android Things working properly. 

I had success with receiving UDP Connection and data without no issues. But no luck on the sender part. It haven't been possible for me to send anything over a LowPAN. 

Anyone have some suggestions. Very much appreciated.

PaulTR, 2019-02-14:
What hardware are you using, and is your code posted anywhere?

nikiizvorski, 2019-02-14:
I am using Android things and the sample lowpan. And just any udp android code its 2 lines. If you want me i will post the exact lines.

TavecchiaG, 2019-02-18:
Hi,
We have the same problem.
Do you have any suggestions?
Best regards,
Giorgio

nikiizvorski, 2019-02-18:
Here is the topic we have raised quys:

https://github.com/androidthings/sample-lowpan/issues/4